miao1007/Openwrt-NetKeeper #157: 运行sh nk4conf.sh时出现错误

LinKey0811, 2017-09-06:
nk4.sh: line 14: syntax error: bad substitution
应该是这一行username="${BASH_REMATCH[1]}"
这个是什么原因呢，不是很熟悉shell的代码

cappuccinoqi, 2017-09-06:
我遇到的错误是第8行：pppoe-server -k -I br-lan，请问你的pppoe版本是3.12的么

LinKey0811, 2017-09-06:
我的是3.11的

cappuccinoqi, 2017-09-06:
我还没有找到适合 mt7620的3.11 版本

LinKey0811, 2017-09-06:
直接运行opkg的代码，就可以了，我是这样操作的

cappuccinoqi, 2017-09-06:
我路由器的源里面的package只有3.12版本的

LinKey0811, 2017-09-06:
我的源是16.02的好像，你看看作者给出的那个源，我不在学校里现在😂

cappuccinoqi, 2017-09-06:
好的

stawidy, 2017-09-07:
@LinKey0811 我用之前的版本是可以正常执行的，最近几天重新刷了下路由器发现这脚本出现了你所描述的错误，我看了下 `BASH_REMATCH[n]` 这个数组是bash的特性，路由器的内置shell应该不是bash，可能不支持这个特性。估计改了脚本那位大兄弟也没在op上跑过。

Zhuhao0922, 2017-09-07:
我也遇到了同样的问题，你解决了吗？我也是3.11的版本

LinKey0811, 2017-09-07:
好吧，后面我也没测试过了😂

Zhuhao0922, 2017-09-07:
@LinKey0811 我刚刚解决了这个问题，你进入我的github，使用那个nk4conf.sh，用的是 @miao1007 的历史版本。现在无线路由器可以用翼讯上网了。

LinKey0811, 2017-09-07:
好😂，改天有空的时候测试一下可行不

stawidy, 2017-09-07:
@Zhuhao0922 你的pppoe-server版本是3.11？似乎LEDE 17.01.2里面的是3.12，跑不起来。

Zhuhao0922, 2017-09-07:
@stawidy 你测试一下吧，应该和pppoe-Server版本关系不大

stawidy, 2017-10-20:
@Zhuhao0922 我找到原因了，LEDE默认的固件没有ipset，运行设置脚本的时候会有warning。