miao1007/Openwrt-NetKeeper #165: 路由器运行sh nk4conf.sh 错误

evindu, 2017-09-14:
运行之后提示
`/etc/rc.common: line 138: ss_tunnel_stop: not found
missing parameters. check config
   ! Failed with exit code 1
killall: apctrl: no process killed
killall: apctrl: no process killed`

完整显示
`[root@PandoraBox:/root]#sh nk4conf.sh
uci: Entry not found
Warning: Section @zone[1] (wan) cannot resolve device of network 'netkeeper'
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Forward 'lan' -> 'wan'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
 * Running script '/usr/share/miniupnpd/firewall.include'
 * Running script '/usr/share/shadowsocks/firewall.include'
/etc/rc.common: line 138: ss_tunnel_stop: not found
missing parameters. check config
killall: apctrl: no process killed
killall: apctrl: no process killed
[root@PandoraBox:/root]#`

Warm-rain, 2017-09-14:
不影响的，你看下接口改好没，服务里面有nk4没，都有就直接登录

evindu, 2017-09-14:
@Warm-rain  进程有 sh nk4.sh  但是pppoe.log里面是空的  我觉得应该没有运行起来。我是小米mini路由器 刷的pandorabox

Warm-rain, 2017-09-15:
@evindu 那就可能是 rp-pppoe-server_服务端的问题

stawidy, 2017-09-21:
你看看rp-pppoe-server的版本，LEDE的3.12不能正常工作，还没有仔细看过。

evindu, 2017-09-21:
换了个版本 搞定了

closed 2017-09-21 by evindu