miao1007/Openwrt-NetKeeper #166: 心跳插件编译问题，头大了哥

RuiKylin, 2017-09-19:
没接触过C，云里雾里。
军训完了网上下单一个K3，上午编译完拨号插件之后发现没南昌的，不知道其他版本是否能用
然后开始弄DaoNet
爬了半天的C编译教程，刚开始无法找到mbedtls/aes.h，于是安装了mbedtls，还是无法找到，
发现在 aes.h 文件在/usr/local/include  下，完全不懂，于是继续爬，修改了DaoNet文件夹下的makefire
将`CFLAGS = -g -Wall -Wimplicit-function-declaration -I $(STAGING_DIR)/include/` 
改为了`CFLAGS = -g -Wall -Wimplicit-function-declaration -I/usr/local/include/`
然后文件找到了，但是之后又提示我引用的SDK中缺少了文件libmbedcrypto.a
但是我下载到的K3 SDK里面的Toolchain文件夹下没有这个文件，文件我在系统的**usr/local/lib**中找到了

我想问问是不是因为我改了makefile造成的，但是不改提示找不到aes.h
搞了一天实在不知道该怎么做了，爬了半天Google没爬出结果了，想不通了Orz

Destinyzxn, 2017-09-24:
我之前编译的时候 也有这个问题 求解答

RuiKylin, 2017-09-24:
@Destinyzxn  
 **libmbedcrypto.a** 是 **mbed TlS** 的 library 文件。

- 你需要先编译一个 **mbedtls**
   GitHub 上直接搜索 mbedtls

- 编译好 **mbedtls** 之后，再进行心跳插件的编译即可。

Destinyzxn, 2017-09-24:
编译之后 放进了lib文件夹内 在daonet目录下make 报错
root@VM-111-115-ubuntu:~/Openwrt-NetKeeper/src/DaoNet/DaoNet# make
mipsel-openwrt-linux-gcc  -o main.o main.c -g -Wall -Wimplicit-function-declaration -I /root/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/include/  -c
In file included from main.c:19:0:
netkeeper.h:15:25: fatal error: mbedtls/aes.h: No such file or directory
 #include <mbedtls/aes.h>
                         ^
compilation terminated.
Makefile:8: recipe for target 'main.o' failed
make: *** [main.o] Error 1
查百度说 安装mbedtls库 我也试着安装了 也还是报错

RuiKylin, 2017-09-24:
你的路由器CPU为MT7620，不需要你自己编译。
项目里有现成的.so文件，至于心跳是江西专用的。
留下的Email通讯OK？

Destinyzxn, 2017-09-24:
814182054@qq.com
so编译我会

closed 2017-09-24 by RuiKylin