miao1007/Openwrt-NetKeeper #167: 日志和接口显示拨号成功但是没有网

XiaoCool007, 2017-09-24:
你好我是南昌大学的，之前用的是K2拨号，插件是在这下载，现在我想换成K3，固件是LEDE，用你的源码编译，由于你说更新了不赞成适用所以我尝试了编译了v32和v18-v29的，在K3上日志显示拨号成功，接口也显示有ipv4地址，但是无法上网，
附上日志：

Tue Aug 22 12:05:19 2017 daemon.notice netifd: Interface 'wan' is now down
Tue Aug 22 12:05:19 2017 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Aug 22 12:05:19 2017 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Tue Aug 22 12:05:19 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 50:8f:4c:52:5e:7b
Tue Aug 22 12:05:19 2017 daemon.notice hostapd: wlan1: AP-DISABLED 
Tue Aug 22 12:05:19 2017 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING 
Tue Aug 22 12:05:19 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Tue Aug 22 12:05:19 2017 daemon.notice netifd: Interface 'lan' is now down
Tue Aug 22 12:05:19 2017 kern.info kernel: [  343.713000] br-lan: port 3(wlan1) entered disabled state
Tue Aug 22 12:05:19 2017 kern.info kernel: [  343.718460] br-lan: port 2(wlan0) entered disabled state
Tue Aug 22 12:05:19 2017 kern.info kernel: [  343.723926] br-lan: port 1(eth0.1) entered disabled state
Tue Aug 22 12:05:19 2017 kern.info kernel: [  343.740641] device wlan1 left promiscuous mode
Tue Aug 22 12:05:19 2017 kern.info kernel: [  343.745237] br-lan: port 3(wlan1) entered disabled state
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: Plugin rp-pppoe.so loaded.
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: Plugin nanchang_sxplugin.so loaded.
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: sxplugin : init
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: -------------------------------------
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: timenow(Hex)=599bad7f
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: Begin : beforeMD5
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 1.<??>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 2.<??6130116086>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 3.<??6130116086jiangxi4.0>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 4.length=<24>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: End : beforeMD5
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: Begin : afterMD5
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 1.MD5use_1=< 7>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: 2.MD5use_2=< d>
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: End : afterMD5
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: -------------------------------------
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: sxplugin : user  is <
[`TUP 7d6130116086@ndcard> 
Tue Aug 22 12:05:19 2017 daemon.info pppd[8404]: sxplugin : passwd loaded
Tue Aug 22 12:05:19 2017 daemon.notice pppd[8404]: pppd 2.4.7 started by root, uid 0
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.191513] device eth0.1 left promiscuous mode
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.196083] device eth0 left promiscuous mode
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.200546] br-lan: port 1(eth0.1) entered disabled state
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.210381] IPv6: ADDRCONF(NETDEV_UP): eth0.1: link is not ready
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.747494] device wlan0 left promiscuous mode
Tue Aug 22 12:05:20 2017 kern.info kernel: [  344.752105] br-lan: port 2(wlan0) entered disabled state
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Interface 'lan' is disabled
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Network device 'wlan1' link is down
Tue Aug 22 12:05:20 2017 daemon.notice netifd: VLAN 'eth0.1' link is down
Tue Aug 22 12:05:20 2017 daemon.notice netifd: bridge 'br-lan' link is down
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Aug 22 12:05:20 2017 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue Aug 22 12:05:20 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
Tue Aug 22 12:05:20 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
Tue Aug 22 12:05:20 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Interface 'loopback' is now down
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Interface 'loopback' is disabled
Tue Aug 22 12:05:20 2017 daemon.notice netifd: radio1 (8415): command failed: No error information (-524)
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Network device 'lo' link is down
Tue Aug 22 12:05:20 2017 daemon.notice netifd: Interface 'loopback' has link connectivity loss
Tue Aug 22 12:05:20 2017 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: No such device
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: tunnel
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: Network device
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: bridge
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: veth
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: macvlan
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: 8021ad
Tue Aug 22 12:05:21 2017 user.notice : Added device handler type: 8021q
Tue Aug 22 12:05:22 2017 kern.info kernel: [  346.676085] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Tue Aug 22 12:05:22 2017 kern.info kernel: [  346.682142] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
Tue Aug 22 12:05:22 2017 kern.info kernel: [  346.688324] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
Tue Aug 22 12:05:22 2017 kern.info kernel: [  346.695761] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Tue Aug 22 12:05:22 2017 kern.info kernel: [  346.702366] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
[... 359 lines cut ...]
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain test
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain onion
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain localhost
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain local
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain invalid
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.net
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.org
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.com
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using 3 more local addresses
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: reading /tmp/resolv.conf.auto
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain test
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain onion
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain localhost
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain local
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain invalid
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.net
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.org
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using local addresses only for domain example.com
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using nameserver 202.101.224.69#53
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using nameserver 202.101.226.69#53
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: using 3 more local addresses
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: read /etc/hosts - 5 addresses
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq-dhcp[13827]: read /etc/ethers - 0 addresses
Tue Aug 22 12:07:58 2017 daemon.info dnsmasq[13827]: exiting on receipt of SIGTERM

XiaoCool007, 2017-09-25:
大佬帮忙看下呗，我看下了下mipsel用的拨号插件应该是V32的，我编译的arm版不知道为什么就有问题

StchaoL, 2017-09-27:
我说几句个人的看法哈。
你这个日志显示pppoe认证成功，并且接口也已经up了，也就是说拨号成功了。但是既然没有网络，你可以试试路由器是否ping的通百度或者114 dns，我猜测可能是防火墙配置错误导致lan口不能通过netkeeper接口连接网络。

sunflyer, 2017-10-06:
江西V32应该早就GG了

XiaoCool007, 2017-10-07:
Miao007直接提供的那个还是可以用的

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

miao1007, 2017-10-11:
可能是防火墙，跃点，DNS等设置错误,你的pppoe是ok的

XiaoCool007, 2017-10-13:
额，不重要了，我把k3卖了，换了k2p

发送自我的 Windows 10 手机