miao1007/Openwrt-NetKeeper #168: 大神求教山东netkeeper更新到4.11，我只推出了组合是\r19^2<^ b2+账号，但是不会写so文件，求教大神帮写一个so文件。

cammer233, 2017-09-25:
大神求教山东netkeeper更新到4.11，我只推出了组合是\r19^2<^ b2+账号，但是不会写so文件，求教大神帮写一个so文件。

huipengly, 2017-09-25:
**怎么推出组合的啊？**
你知道账号的话不需要so文件就可以拨号。在openwrt里面运行下面这两句，可以支持转义字符。然后直接新建pppoe拨号就可以。
```sh
cp /lib/netifd/proto/ppp.sh /lib/netifd/proto/ppp.sh_bak
sed -i '/proto_run_command/i username=`echo -e "$username"`' /lib/netifd/proto/ppp.sh
```

cammer233, 2017-09-25:
是用客户端拨号和路由器n4k.sh那个文件截取的，之前使用的so的文件拨号的，感觉方便一些，so文件容易写吗。

huipengly, 2017-09-25:
。。。。。。。。。。。。。。。。。。。。。。。。。。。
1.账号是会变的，不是可以一直用的。
2.不知道账号变化的规律。
3.所以用脚本截取。

huipengly, 2017-09-25:
**2.不知道账号变化的规律。**
3.所以用脚本截取。

我也是闲着蛋疼才回你。

cammer233, 2017-09-25:
还是感谢，等大神有解决方案。。

StchaoL, 2017-09-27:
山东新版的是去掉了后缀，也就是RADIUS=“”
编译的时候指定一下就可以了。
虽然我也在卖破解路由器，但是还是希望有更多人能自己动手，丰衣足食。

cammer233, 2017-09-28:
这个新版说的是4.9是4.11？

jth445600, 2017-10-31:
@cammer233 能不能给我说一下，你是怎么弄的，我想试试，可是我看不懂那个readme文件

closed 2018-04-22 by Mr-cuiks