miao1007/Openwrt-NetKeeper #169: 建议以后对武昌的不再做支持。

873578156, 2017-10-01:
卖一个路由器300块，都卖到山东了，在我朋友的学校群里发广告。还说是自己破解的，qq定位的ip是湖北武昌。

huipengly, 2017-10-01:
我觉得吧，有支持至少还可以自力更生。

873578156, 2017-10-01:
也许吧

Tomsu2016, 2017-10-01:
其实我很好奇，为什么能卖到300块？
根据gayhub评论的逻辑：我的东西不能用来赚钱（被人发现了要坐牢？），所以我不出代码。
根据市场经济的逻辑：物以稀为贵，会用的人很少，作者也不出代码，带心跳的路由器价钱水涨船高。
当然作者和卖家什么关系就不清楚了。

Tomsu2016, 2017-10-01:
哦，对了。以后估计这代码也卖不出去钱了。
因为湖北的电信校园网涨价了，另外移动的网（直接给宽带账号）质量也起来了，e信可能只是这几届学长们的回忆了。

873578156, 2017-10-01:
我也是偶然在我朋友的学校的新生群看到

873578156, 2017-10-01:
@Tomsu2016 我和卖家没关系，我只是一个正在成长的码农

Tomsu2016, 2017-10-01:
楼主和作者？？？
sunflyer的捐赠授权，你可以去https://simplenetkeeper.com/看看了解下，反正我是没看出来这个捐赠和购买有什么区别。

873578156, 2017-10-01:
@Tomsu2016 简直666了，

sunflyer, 2017-10-06:
@873578156 湖北电信我已经不太想再跟了，新版心跳基于HTTP
我只说我现在知道的东西：
服务器 hbxt.xinli.com.cn:10089
heartkey  自己找，很简单
签名算法 HMAC-SHA1
签名参考标准应该是 RFC2104

至于账号算法，还是2.5，换了PIN而已，关键字timepin

@huipengly 作为之前的参与者，现在的我可能没有资格说这句话，但还是要说一句：非常感谢你对这个repo的贡献。

@Tomsu2016 随你自己的定义：）你要当作购买也不是我能改变的事情。如果真的一定要让我要找一个借口的话就是，购买要提供售后，然而捐赠不会。何况这软件我也确实没有所谓的“售后”。如果非要想着用十几块钱就能帮这群人解决全部问题的话，我也没这个时间。

有关我软件对特定地区强制“捐赠”的问题，为什么重庆陕西山东电信都不要捐赠，唯独湖北和山东移动需要？你们爱说我什么都是你们的自由，但是我必须要说明我的立场，总结下来一句：**你们对作为开发者的我有多少尊重，我对你们就有多少尊重。**

首先，早前2.5版本开源的时候，我做完了几乎全部的事情，路由器适配也好，其他几个地区的Radius拆解也好，基本都是从大佬那边知道原理以后弄过来我编完了分发出去的。我没有拿开源的东西去“挣取”、“索要”的这个资本——连大佬都没这么做，我这么做未免也太说不过去。

然而，免费分发的一个问题就是：某些人拿到软件以后，发现这软件有利可图，然后到处坑人。
我当时做完APP以后不到一周，星空极速吧里面就有人发帖出售，30一份，你回贴说他拿你的做的东西卖，他还会叫嚣“老子就是卖钱了你他吗个煞笔做完了不给我卖你算老几”——虽然发完了就删掉了，不过我还是记得非常清楚的，因为这是第一个叫嚣的。

然后，即便是我声明了“卖软件的司马”，然而2015年下半年到2016年上半年时间，魔改我APP、倒卖我软件的事情不胜枚举，最典型的就是东华理工长江学院的“威创科技”（还是叫什么鸟名字来着）/湖北某学校学生的“Simple e信”/山东青岛琴岛学院的“琴岛逆风”，一个个玩魔改玩的贼欢快。

在那个时候我所了解的倒卖最严重的地区是江西，甚至当时还和江西科技学院的尼古丁撕了一笔，这个当时很多人都知道的，从这件事开始持续了相当一段长时间我对江西的同学没有任何好感（直到湖北这帮子事情发生以后我才发现原来江西的同学都是好人）。在那之后没多久江西电信开始更新4.X，换了算法，当时做的思路是从其他手段拦截账号。事件查看器当时可以抓到真实帐号，我当时还心里还在想说我这是不是做得太绝了，本身软件放出来就是方便你们用的，江西的同学没路由器用了怕不是个事，然后就做了一个手动版本——自己从事件查看器复制账号然后设置。然后呢，这个手动版本发在群里第二天就被人拿出去卖，售价60。最搞笑的是，软件出了问题以后，那个卖软件的直接说了一句“有问题加这个群XXXX”（群就是我当时拿来帮助解决问题的QQ群）。然后随着这软件的铺开，信利在4.8彻底堵住了事件查看器这个洞。

后面的事情就不多说了，全国各地陆续开始4.X算法部署（除开湖北电信），和新版本客户端升级，这时候就慢慢暴漏出来，山东移动更新完以后有一个人跑到我博客问“怎么琴岛逆风不能逆了啊？”，我当时还不知道什么这特么是个鬼东西，直到几天之后群里面有一个琴岛学院的人说他们有一个手机APP设置路由器的就叫这名字，发过来一看，合着拿了老子的源代码就改了个名字，其他的什么都没改。

至于湖北？只说一个，你翻下这个Repo的issue历史，有一个标题为省略号的，点开看看就知道原因了。

如果只有JS都还好，最可气的是，某些人作为大学生一点脑子都没有。除了伸手，还是伸手。懒到连百度都不会，现成的文档都不肯看。白底黑字告诉你什么问题了，他还要问这是什么意思。你不帮他，他还一脸“你他吗的帮下会死啊”的表情。
我不相信一个到了大学的人连阅读理解都不会，说的不好听一点就是又蠢又懒。这也正好是这帮卖路由器的人的目标对象。

从那以后开始我觉得我没法接受这个国家某些人的行为。**我作为一个开发者忙死忙活想办法解决问题，换来的就是一帮小白在这里不动脑子的狂热伸手，以及这帮小人在这里坑人还他妈把锅往我这里甩。**

我作为一个独立开发者，实在不喜欢某些人拿着我免费的东西在外面坑了人以后，还要把所有出现的问题全部甩给我的这种情况。十几块钱，甚至于最近某些人一直批判的湖北电信的45块钱，我说白了都只是为了作为一个门槛。

**你们对我没有尊重，我就让你们强行从物质上尊重。**
你们怎么批判是你们的事情。

Warm-rain, 2017-10-07:
@sunflyer +1

XiaoCool007, 2017-10-07:
我只想默默说句，既然放出代码，用来分享了而不是获取收益，那么你为什么还要对于开源的东西死抓不放，何况他既然卖那么他也是要付出时间，卖家和买家一个愿打一个愿挨，买家愿意花钱而不花脑子你有什么办法，更何况，实际上很多人上大学之前都没怎么碰过电脑，你手把手教他他都不一定会，有些SB愿意花钱你有什么办法

huipengly, 2017-10-07:
@XiaoCool007 

你说有些伸手党sb这个我同意。但是

> 何况他既然卖那么他也是要付出时间 by @XiaoCool007 

小偷偷别人东西也是要付出时间的哦。如果你不认为js是小偷，那我也不说什么了。毕竟开源源码改一改就“自主创新”了。

而且我觉得事情的重点是被恶心到了。等开发者都被恶心走了，祈求后面还有人继续填坑。~反正我脱离电信大坑了~。

Destinyzxn, 2017-10-08:
开源的东西被拿去卖 最后导致的只能是没人开源 没人再分享了

qiang125, 2017-10-08:
就算是开源，也不是人人都会用，学校可能一大半的人都不会用，就算是计算机专业，一半的人可能都没用过github。开源不代表免费，就好比重装系统的教程网上到处都有，可是还是可以收费。给别人装路由器是可以收取部分费用的，没有利益谁会跑到陌生人那里帮他做事？只不过这个价格不能太高，太高了就是坑校友了，也不建议挂在闲鱼淘宝上卖给外校的。这种靠卖路由器为生的多数奸商，有人有这个需求的时候再去弄就好，而不是到处打广告卖路由器。

3370078239, 2017-10-08:
@sunflyer 好像又换了，，我前两天抓是这个，今天抓又变了，，可以教一下怎么写执行脚本吗，谢谢

XiaoCool007, 2017-10-08:
@huipengly 你说的这些人我知道，然而你有什么办法，在国内诸如各个播放器不是基本都用的ffmpeg，比如阿里鸨所谓的第三大操作系统

XiaoCool007, 2017-10-08:
@qiang125 我也是这个看法

gmfghp, 2017-10-09:
我感觉吧，靠自己本事吃饭没问题。但是，自己没实力，剽窃别人的 这就像过街老鼠。楼上说一台路由器卖到300？真的惊到了。但是想想学校为什么限制一人一号？到头来 推动事情发展的就是利益，一个几兆带宽的破网，收了学生多少钱。最后坑的还是学生。而我们这里就是反抗起义的领头人。

3370078239, 2017-10-09:
确实，本来就坑，本来这个是拿来方便学生。。到最后却成为他们赚钱的工具，这和那个信利公司有什么区别

left906080850, 2017-10-10:
@sunflyer 大佬。这群人没办法，屡次刷新三观

sunflyer, 2017-10-10:
@left906080850 你这个时候出来说这句话怕是没什么意思吧

left906080850, 2017-10-10:
@sunflyer 我是昨天才得到消息的，你这个每次都是长篇大论，也是可以

sunflyer, 2017-10-10:
@left906080850  我算是知道湖北这些人有多恶心了。没道理可言。不过也是，13亿人没几个渣子还真以为活在天堂了

Tomsu2016, 2017-10-10:
js哪里都有，只是湖北缺少真正愿意维护的开发者而已。

sunflyer, 2017-10-10:
@Tomsu2016 请你注意我说的是“湖北这些人”，没有一棒子打死你们全部人，我地图炮了吗？

就这些人的所作所为，我还有什么话说？

湖北定制版软件被你们湖北某学校的学生给破解了限制，光明正大去掉APP背景图片去掉作者信息然后打个补丁就拿出去卖60一个，你好意思跟我说“尊重”？
干不了电信，干不了信利，拿我软件开刀，我昨晚说我不做了以后，接着就给我服务器打了11G的DDOS流量，好意思谈尊重？

sunflyer, 2017-10-10:
另外我说一句，这个Repo里面，愿意维护的湖北的开发者，一只手数不过3.
JS确实哪里都有，但是卖路由器强行200以上，卖200以下还会被其他卖路由器的威胁的，仅此你们湖北电信一处。说句不好听的这可以算是你们湖北的特色。
江西去年再怎么凶残，也最多就拿着免费的在外面搞，但是最起码不会有这种组团抬高路由器价格强行坑人的，更不至于破解我或者向左的捐赠版本然后改头换面“自主研发”在外面卖。
你跟我说这不是你们湖北有些人想钱想疯了老子绝对不信。自己回头看看你们湖北电信更新的几个时间点的时候你们的JS到底做了什么。

2.5升级4.X -》 +60块升级
4.X开心跳 -》 +60块升级
4.X换心跳-》 魔改我软件，60块倒卖

非要有人当坏人的话，这个地图炮我就开定了。我不是针对你们湖北某一个奸商，更不针对某一个学校，我是说湖北所有的奸商，都是垃圾。

left906080850, 2017-10-10:
江西一开始这样搞，湖北这样搞又一波，下一波 是哪个地区。
你们魔改就算了，不要说“自主研发” 好不好，我记得没错这个咱们云心跳也被魔改啊。20/月(真是是疯了？)

更新一个二个就像恶狗一样！

有本事魔改客户端！

走好不送，此处MLGB说了。

luodongdong99, 2017-10-11:
唉。。。说到底还是学校垄断，我是武昌这边学校的，宽带费用强制和卡绑定，79一个月，很多女生就买300多的路由器一个宿舍用。:-(

cappuccinoqi, 2017-10-13:
哎 为大佬感到一丝悲哀，两年前我是怎么也想不到如今的结果。本来是为了方便自己使用的东西，流到奸商手里就成了赚钱的工具了，出了问题还甩锅给你，真心寒

Warm-rain, 2017-10-13:
不尊重作者成果到最后受伤的只能是你们自己，不是好人越来越少，而是某些人一次一次的触碰别人的底线。

lee86, 2017-10-19:
讲道理，我也是武汉的，但真的看不下去，这边卖路由的和拿2.x的so骗人的都特么疯了，别人学生的600甚至800骗了很有成就？心里没点愧疚的？学生的钱怎么来的心里没个逼数的？用这个的除了对这玩意儿有兴趣的，还有些人是有特殊用途的，至于学生群体就不可能有很多经济来源，不骗人的能用的我也不说什么了，骗人的真的是该司马，别人学生的钱是拿来干嘛的？九月有天群里有个同是武汉的小兄弟找我问能不能拨号，我也是一个伸手党，我只能跟他说外面的卖这玩意儿的最好不要信，没有售后也就算了，坑学生，坑开发者真的是6得一笔啊

lee86, 2017-10-19:
还有…………我们学校电信果断绑卡销售了，弃坑弃坑，联通大法好。电信卡，不可能的😏

moust33700, 2017-10-19:
@ lee86 600 ，800确实恐怖了啊，我就见过300差不多的。。真希望早点脱坑啊，学校配合电信垄断太难受了，唉

lee86, 2017-10-20:
@moust33700我们学校校园网还是电信资助的，当初是为了黑裙方便才找的so来拨号，电信直接给公网ip，在加上梅林固件的阿里云动态ip插件，很好用，现在把黑裙玩成爱快路由了，加上武汉改协议，电信绑定销售，就决定弃坑了:grin:

xtsb1998, 2017-10-20:
反正呢，每次特别对待都觉得不怎么好受，既然是湖北的某些人。为什么非强调湖北呢，这些 某些人自己心里清楚是说他们，强不强调湖北都是一样的，他们不觉得怎么样的，话就算把名字挂出来也没啥用

cuyl, 2017-12-29:
离开学校一年多，没想到发生这么多事，学弟们真会搞事

Droid-MAX, 2018-10-15:
真不想长大，烦恼多多的未来...

miao1007, 2018-10-15:
> 关于用开源项目挣钱
* 这玩意挣不了大钱，接一个软件项目就够4年宽带了，在学校多学技术后才能挣钱
* 个人鼓励合理付费提供有偿技术支持的形式，既可以获得收益，也可以**提高自己的沟通与技术能力**，最终回馈项目改进。国外的Elastic，RedHat均是这样驱动的
* 利用信息差骗钱是短视行为，毕业后干啥呢，是当卖房中介吃盒饭吗，是做假简历面试被拆穿吗？

> 关于懒惰的大学生
* 这是人家自己的选择，自生自灭。但是说不定毕业后家里就准备好了一切，你不能走偏了也跟着玩

> 关于技术与能力
* 能够掌握其中的断点分析与调试方法，甚至OP的交叉编译，以及比较新的Puppeteer模拟等技术
* 提高自己层次，多学英文

Droid-MAX, 2018-10-16:
大佬们现在是在考研还是在外工作？带带我这个菜鸡吧~

xyl23333, 2020-09-09:
这种情况还是比较常见的，我们浙江的学校也有这么恶劣的情况，一台斐讯k2卖到了250，可惜还没提醒寝室长，寝室长已经买了