miao1007/Openwrt-NetKeeper #182: waiting for PADO timeout(服务器无响应错误)

StchaoL, 2017-10-26:
最近遇到好几个路由器都出这个错误, 要嘛是**等待PADO包超时**, 要嘛是**等待PADS包超时**. 但是同样的路由器配置和固件包, 到了别的宿舍楼就不会出问题, 不知道有谁遇到过这种问题吗??? 前几天还碰到个拨号成功后自动掉线的(本校无心跳包), 真真真是诡异........

huipengly, 2017-10-29:
这个错误，我当时在学校的经验是，路由器多次错误拨号后，服务器会暂时封禁netkeeper端口所对应的mac，可以选择先关闭端口，等五分钟再拨号；也可以在/etc/config/network里修改端口mac，随便改一位就可以。

也遇到过这两种方法都试了，但还是无法拨号的情况。

StchaoL, 2017-10-29:
的确, 我也试过修改MAC, 只是偶尔能成功.

我试过使用 Ubuntu抓路由器PPPoE回话时的包, 发现即使是指定了 ``` option macaddr ``` ,路由器发出的PPPoE回话包的Source address依然是修改之前的, 这个问题真的是让我很费解.
后来发现使用macvlan驱动建立虚拟网卡可以解决这个问题, 然后也发现了之前遇到的"自动掉线问题"是因为使用的潘多拉的 ```luci-app-sychdial ``` 自带的"断线检测"在作怪. 

这学期学校的移动宽带真的好费事, 上学期什么问题没有, 这学期就总是"服务器无响应".

miao1007, 2017-11-02:
https://github.com/miao1007/Openwrt-NetKeeper/wiki/%E5%B8%B8%E8%A7%81Log%E6%8A%A5%E9%94%99

closed 2017-11-12 by StchaoL