miao1007/Openwrt-NetKeeper #184: 今天帮别人装插件 别人连不上我自己断开重连结果也连不上了

baoziii, 2017-11-06:
坐标温州(浙江)，没有心跳 断开前已经连上了10多个小时了 一直可以用，用python拨号器也可以连接。路由系统pandora用了两年了 一直没事儿 附上Log 顺便问下“CHAP authentication failed: 109026006|109020131|The subscriber doesn't have the access permission.”是什么情况

`Mon Nov  6 21:50:41 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:41 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:41 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:41 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:41 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: started, version 2.71 cachesize 150
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq-dhcp[2293]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: using local addresses only for domain lan
Mon Nov  6 21:50:41 2017 daemon.warn dnsmasq[2293]: no servers found in /tmp/resolv.conf.auto, will retry
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: read /etc/hosts - 1 addresses
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: read /tmp/hosts/dhcp - 1 addresses
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq[2293]: read /tmp/hosts/odhcpd - 0 addresses
Mon Nov  6 21:50:41 2017 daemon.info dnsmasq-dhcp[2293]: read /etc/ethers - 0 addresses
Mon Nov  6 21:50:41 2017 user.notice upnp daemon: external interface not found, not starting
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:42 2017 daemon.warn odhcpd[1446]: DHCPV6 SOLICIT IA_NA from 00010001213da67bd8cb8ac58021 on br-lan: no addresses available 
Mon Nov  6 21:50:43 2017 daemon.notice pppd[1602]: Connection terminated.
Mon Nov  6 21:50:43 2017 daemon.notice pppd[1602]: Modem hangup
Mon Nov  6 21:50:43 2017 daemon.info pppd[1602]: Exit.
Mon Nov  6 21:50:43 2017 daemon.notice netifd: Interface 'wansx' is now down
Mon Nov  6 21:50:43 2017 daemon.notice netifd: Interface 'wansx' is disabled
Mon Nov  6 21:50:43 2017 daemon.notice netifd: Interface 'wansx' is enabled
Mon Nov  6 21:50:43 2017 daemon.notice netifd: Interface 'wansx' is setting up now
Mon Nov  6 21:50:43 2017 daemon.notice netifd: wansx (2455): uci: Entry not found
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: Plugin rp-pppoe.so loaded.
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: Plugin sxplugin.so loaded.
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: sxplugin : init
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: -------------------------------------
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: timenow(Hex)=%ßx
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: Begin : beforeMD5
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 1.<>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 2.<>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 3.<>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 4.length=<1>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: End : beforeMD5
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: Begin : afterMD5
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 1.MD5use_1=< a>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: 2.MD5use_2=< 8>
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: End : afterMD5
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: -------------------------------------
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: sxplugin : user  is <
44)A0 a8*******************> 
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: sxplugin : passwd loaded
Mon Nov  6 21:50:44 2017 daemon.notice pppd[2477]: pppd 2.4.7 started by root, uid 0
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: PPP session is 32839
Mon Nov  6 21:50:44 2017 daemon.warn pppd[2477]: Connected to 00:18:82:70:4b:9e via interface eth0.3
Mon Nov  6 21:50:44 2017 daemon.info pppd[2477]: Using interface pppoe-wansx
Mon Nov  6 21:50:44 2017 daemon.notice pppd[2477]: Connect: pppoe-wansx <--> eth0.3
Mon Nov  6 21:50:47 2017 daemon.info pppd[2477]: syncppp not active
Mon Nov  6 21:50:47 2017 daemon.info pppd[2477]: CHAP authentication failed: 109026006|109020131|The subscriber doesn't have the access permission.
Mon Nov  6 21:50:47 2017 daemon.err pppd[2477]: CHAP authentication failed
Mon Nov  6 21:50:54 2017 daemon.notice pppd[2477]: Connection terminated.
Mon Nov  6 21:50:54 2017 daemon.notice pppd[2477]: Modem hangup
Mon Nov  6 21:50:54 2017 daemon.info pppd[2477]: Exit.
Mon Nov  6 21:50:54 2017 daemon.notice netifd: Interface 'wansx' is now down
Mon Nov  6 21:50:54 2017 daemon.notice netifd: Interface 'wansx' is disabled
Mon Nov  6 21:50:54 2017 daemon.notice netifd: Interface 'wansx' is enabled
Mon Nov  6 21:50:54 2017 daemon.notice netifd: Interface 'wansx' is setting up now
Mon Nov  6 21:50:54 2017 daemon.notice netifd: wansx (2528): uci: Entry not found
Mon Nov  6 21:50:54 2017 daemon.info pppd[2550]: Plugin rp-pppoe.so loaded.
Mon Nov  6 21:50:54 2017 daemon.info pppd[2550]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Mon Nov  6 21:50:54 2017 daemon.info pppd[2550]: Plugin sxplugin.so loaded.
[... 145 lines cut ...]
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: sxplugin : init
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: -------------------------------------
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: timenow(Hex)=%ßx
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: Begin : beforeMD5
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 1.<>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 2.<>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 3.<>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 4.length=<1>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: End : beforeMD5
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: Begin : afterMD5
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 1.MD5use_1=< a>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: 2.MD5use_2=< 8>
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: End : afterMD5
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: -------------------------------------
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: sxplugin : user  is <
4<)AQ a8*******************> 
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: sxplugin : passwd loaded
Mon Nov  6 21:51:33 2017 daemon.notice pppd[2832]: pppd 2.4.7 started by root, uid 0
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: PPP session is 32529
Mon Nov  6 21:51:33 2017 daemon.warn pppd[2832]: Connected to 00:18:82:70:4b:9e via interface eth0.3
Mon Nov  6 21:51:33 2017 daemon.info pppd[2832]: Using interface pppoe-wansx
Mon Nov  6 21:51:33 2017 daemon.notice pppd[2832]: Connect: pppoe-wansx <--> eth0.3
Mon Nov  6 21:51:36 2017 daemon.info pppd[2832]: syncppp not active
Mon Nov  6 21:51:36 2017 daemon.info pppd[2832]: CHAP authentication failed: 109026006|109020131|The subscriber doesn't have the access permission.
Mon Nov  6 21:51:36 2017 daemon.err pppd[2832]: CHAP authentication failed`

baoziii, 2017-11-06:
终于连上了
``
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: -------------------------------------
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: timenow(Hex)=%ßx
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 3.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 4.length=<1>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : afterMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.MD5use_1=< a>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.MD5use_2=< 8>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : afterMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: -------------------------------------
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: sxplugin : user  is <
M4)AQ a8********************> 
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: sxplugin : passwd loaded
Mon Nov  6 22:11:27 2017 daemon.notice pppd[7640]: pppd 2.4.7 started by root, uid 0
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: PPP session is 31714
Mon Nov  6 22:11:27 2017 daemon.warn pppd[7640]: Connected to 00:18:82:70:4b:9e via interface eth0.3
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Using interface pppoe-wansx
Mon Nov  6 22:11:27 2017 daemon.notice pppd[7640]: Connect: pppoe-wansx <--> eth0.3
Mon Nov  6 22:11:30 2017 daemon.info pppd[7640]: syncppp not active
Mon Nov  6 22:11:30 2017 daemon.info pppd[7640]: CHAP authentication succeeded: Authentication success,Welcome!
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: CHAP authentication succeeded
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: peer from calling number 00:18:82:70:4B:9E authorized
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: local  IP address 122.239.218.***
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: remote IP address 122.239.216.1
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: primary   DNS address 61.153.177.196
Mon Nov  6 22:11:30 2017 daemon.notice pppd[7640]: secondary DNS address 61.153.177.197
Mon Nov  6 22:11:30 2017 daemon.notice netifd: Network device 'pppoe-wansx' link is up
Mon Nov  6 22:11:30 2017 daemon.notice netifd: Interface 'wansx' is now up
Mon Nov  6 22:11:30 2017 user.notice firewall: Reloading firewall due to ifup of wansx (pppoe-wansx)
Mon Nov  6 22:11:31 2017 user.notice ddns-scripts-myddns: Running IP check ...
Mon Nov  6 22:11:32 2017 daemon.info dnsmasq[2293]: reading /tmp/resolv.conf.auto
Mon Nov  6 22:11:32 2017 daemon.info dnsmasq[2293]: using local addresses only for domain lan
Mon Nov  6 22:11:32 2017 daemon.info dnsmasq[2293]: using nameserver 61.153.177.196#53
Mon Nov  6 22:11:32 2017 daemon.info dnsmasq[2293]: using nameserver 61.153.177.197#53
Mon Nov  6 22:11:32 2017 daemon.err miniupnpd[7973]: could not open lease file: /var/upnp.leases
Mon Nov  6 22:11:32 2017 daemon.notice miniupnpd[7973]: HTTP listening on port 5000
Mon Nov  6 22:11:32 2017 daemon.warn miniupnpd[7973]: no HTTP IPv6 address, disabling IPv6
Mon Nov  6 22:11:32 2017 daemon.notice miniupnpd[7973]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Nov  6 22:11:33 2017 user.notice ddns-scripts-myddns: Update successful

kuretru, 2017-11-07:
今天同样发生此问题，注意观察日志
```
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: timenow(Hex)=%ßx
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 3.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 4.length=<1>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : afterMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.MD5use_1=< a>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.MD5use_2=< 8>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : afterMD5
```
这是正常拨号成功的日志
```
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: -------------------------------------
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: timenow(Hex)=59ffef50
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: Begin : beforeMD5
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 1.<??>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 2.<??xxxxxxxxxxx>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 3.<??xxxxxxxxxxxsinglenet01>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 4.length=<26>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: End : beforeMD5
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: Begin : afterMD5
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 1.MD5use_1=< 6>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: 2.MD5use_2=< 9>
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: End : afterMD5
Mon Nov  6 13:12:48 2017 daemon.info pppd[15131]: -------------------------------------
```
两者中间可以很明显的发现1-3没有值赋进去，猜测是对时间处理产生了异常，将时间改回昨天6号之前就可以拨号成功

kuretru, 2017-11-08:
错误原因具体可见issue#188，内有临时解决办法，我已提交一个pull request等待作者merge。