miao1007/Openwrt-NetKeeper #187: 路由器一重启就无法连接了

NeuraxLin, 2017-11-08:
RT 每次都要恢复出厂再重新进行sh 之后只有第一次正常 页面点击重启之后就无法连接
错误的Log大概是这样
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: timenow(Hex)=xxxxxxxx（正常的）
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Begin : beforeMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 1.<>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 2.<>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 3.<>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 4.length=<27>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: End : beforeMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Begin : afterMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 1.MD5use_1=< d>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 2.MD5use_2=< 4>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: End : afterMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
就是BeforeMD5之后的三个值计算不出来
间接导致连接不上
但是第一次破解后重启 同样的文件 一切正常
进行页面重启或者断电就无法计算了
陕西电信翼讯 无心跳 小米Mini

NeuraxLin, 2017-11-08:
是时间赋值错误了吗 好像是最近的问题

kuretru, 2017-11-08:
错误原因具体可见issue#188，内有临时解决办法，我已提交一个pull request等待作者merge。

957313677, 2018-08-30:
陕西电信用哪个so文件呀！

957313677, 2018-08-30:
求个so文件

NeuraxLin, 2018-08-30:
@957313677 给个邮箱

957313677, 2018-08-30:
@NeuraxLin

957313677, 2018-08-31:
@NeuraxLin 不在吗？？？

NeuraxLin, 2018-08-31:
@957313677 还没开学 so在学校的笔记本里 等明天咯

957313677, 2018-08-31:
@NeuraxLin 给个联系方式吧！

NeuraxLin, 2018-08-31:
@957313677 lby123@vip.qq.com