miao1007/Openwrt-NetKeeper #188: 时间戳位移后为0，造成的错误字符串长度问题

kuretru, 2017-11-08:
```
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: timenow(Hex)=%ßx
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 3.<>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 4.length=<1>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : beforeMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: Begin : afterMD5
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 1.MD5use_1=< a>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: 2.MD5use_2=< 8>
Mon Nov  6 22:11:27 2017 daemon.info pppd[7640]: End : afterMD5
```
近日出现拨号不成功的情况，可以看到传入MD5进行计算的字符串长度为0，原因为时间戳进行位移运算后的结果类似于`0x12 0x00 0x8f 0x38`产生了0，导致后续算法`strlen(beforeMD5)`取到得值为0，传入MD5进行计算的字符串不正确。

可以将路由器时间更改为11月6号之前，临时解决该问题。

zuishikonghuan, 2017-11-10:
万分感谢，我把这个项目移植到 Debian 后，去掉了调试信息，拨不上号了还以为是算法更新了，感谢！

ALazyLeaner, 2017-11-10:
@kuretru 能问下，我自己对着教程编译了从你那下载的代码，在路由器里拨号时显示CHAP认证成功，但是就是没网是怎么回事呢，我这是江西南昌，这个方式还是可以破解的，计算账号时显示jiangxi4.0，但是就如你所说的，从6号之后就经常连不上，只有改日期才行，不知道能否帮下忙，垃圾宽带实在闹心

closed 2018-08-30 by kuretru