miao1007/Openwrt-NetKeeper #192: 一点点奇葩思路

LittleWhite-Carry, 2017-11-16:
刚入坑，就发现重庆要更新？

先说一个新发现，如果你有动态密码和固定密码两种登录方式，而且两种方式账号不一样，那么恭喜你这两种方式同时拨号可以登录，网速叠加。

我不懂什么网络层或者什么协议，前段时间才把pppoe搞懂一点点，改了改大佬的文件，配合自己的python写了一个动态多拨网速叠加的程序，分享给重庆的同学们最后狂欢一下。没解决心跳问题。

好下面说说思路吧。重庆我们学校有网页认证，有没有办法把网页认证的包从电脑传给路由器然后让路由器连接上？然后就是之前看到过的我自己也想过的，本地发送心跳到底怎么搞？有没有大佬能搞出来，我可以整合一下到我的软件里面，也方便大家。

最后有没有大佬想搞心跳，但是没时间的，也可以带带我，我自愿接盘。不为赚钱，只是想搞搞新东西。直到我研究生毕业。今年刚刚研一，有的话期望大佬能回复我。然后带带我。本身不太懂协议什么的，但是编程还是可以的。

有技术也有志愿一起搞的也可以留言。

yhm97, 2017-12-17:
大佬能不能解决下进内网的问题
之前我单播直接用route add在路由器里面添加路由就能同时进内网和外网
现在多播了添加路由还是进不去内网
不知道怎么弄，现在只能断了网进内网

LittleWhite-Carry, 2017-12-18:
@yhm97 其实是个伪大佬，不怎么懂这些东西。你说的问题，可能可以从负载均衡下手，因为联想到之前的https在多拨的时候不稳定，会出现ip飘逸的现象，所以所有的https协议都走了一个端口，是否你这个进入内网也会有这样的问题，或者跟这个类似，你可以去考虑一下。