miao1007/Openwrt-NetKeeper #198: 南昌地区 拨号失败问题 unrecognized option 'nanchang_sxplugin.so'

shana321, 2018-01-04:
Thu Jan  4 12:02:56 2018 daemon.err pppd[11647]: unrecognized option 'nanchang_sxplugin.so'
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): /usr/sbin/pppd: unrecognized option 'nanchang_sxplugin.so'
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): pppd version 2.4.7
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): Usage: /usr/sbin/pppd [ options ], where options are:
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	<device>	Communicate over the named device
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	<speed>		Set the baud rate to <speed>
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	<loc>:<rem>	Set the local and/or remote interface IP
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 			addresses.  Either one may be omitted.
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	asyncmap <n>	Set the desired async map to hex <n>
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	auth		Require authentication from peer
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647):         connect <p>     Invoke shell command <p> to set up the serial line
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	crtscts		Use hardware RTS/CTS flow control
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	defaultroute	Add default route through interface
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	file <f>	Take options from file <f>
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	modem		Use modem control lines
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): 	mru <n>		Set MRU value to <n> for negotiation
Thu Jan  4 12:02:56 2018 daemon.notice netifd: netkeeper (11647): See pppd(8) for more options.
Thu Jan  4 12:02:56 2018 daemon.notice netifd: Interface 'netkeeper' is now down
Thu Jan  4 12:02:56 2018 daemon.notice netifd: Interface 'netkeeper' is disabled
Thu Jan  4 12:02:56 2018 daemon.notice netifd: VLAN 'eth0.2' link is down
Thu Jan  4 12:02:56 2018 daemon.notice netifd: Interface 'netkeeper' has link connectivity loss

shana321, 2018-01-04:
已经上传过SO文件了，路径也没问题，拨号日志如上

shana321, 2018-01-04:
已解决  sh文件命令错了

closed 2018-01-04 by shana321

YakumoriYukari, 2019-12-06:
请问一下是那个sh文件命令错了？我也遇到了这样的问题