miao1007/Openwrt-NetKeeper #199: 拨号拦截法一个问题，帐号获取是\r拨号变成^M

shana321, 2018-01-05:
[PAP AuthReq id=0x27 user="\r5[8I8& a917346629523@jxcard" password="182010"]
Can't open PAP password file /etc/ppp/pap-secrets: No such file or directory
sent [PAP AuthNak id=0x27 "Login incorrect"]
PAP peer authentication failed for ^M5[8I8& a917346629523@jxcard
sent [LCP TermReq id=0x2 "Authentication failed"]
rcvd [LCP EchoRep id=0x0 magic=0x145d64b7]
rcvd [LCP TermAck id=0x2 "Authentication failed"]
Connection terminated.

huipengly, 2018-01-05:
编辑一下自己的账号密码。别泄漏了。
^m和/r好像是Linux显示的问题。你在网页端看截获的账号有问题么

shana321, 2018-01-05:
谢谢提醒，网页端看的确是/r,那拨号失败是什么原因呢，帐号也成功获取了，路由器web端看IP没被分配到。克隆MAC也试过了没用

huipengly, 2018-01-05:
多拨号几次试试。错误的具体原因我也不清楚

shana321, 2018-01-05:
@huipengly 感谢回答，我再研究研究吧，顺带一提执行脚本后还会出现无法访问路由web端的情况，必须重启后才能访问几次，然后又不能访问了

shana321, 2018-01-05:
SSH连接是可以一直保持的

shana321, 2018-01-05:
@huipengly 谢谢大佬提供的脚本，搞定了。我换成pandorabox之后成功解决，之前用的是恩山abc大神的官改k2固件 拨号不成功，不知道是什么原因。

closed 2018-01-05 by shana321