miao1007/Openwrt-NetKeeper #202: 有么有好心大佬给一个so的编译教程

aojie654, 2018-03-01:
恩山无限论坛里已经有人做出了重庆地区的4.X拨号so，但是其他地区需要从本repo里面自行编译，不知道这个编译到底怎么弄了？有一点点ubuntu使用基础，计算机专业（但是学的很渣……），地区so可以自己搞定么？

aojie654, 2018-03-01:
我是真的想不通买路由器的那些人都是怎么想的，卖单地区的就算了，吧所有2.5协议的so都放进路由器里自己选择，名字好像是个diy-box还是啥？这是害怕所有的2.5协议还没封杀干净？

aojie654, 2018-03-01:
淘宝diybox一搜一大片，还都是四川的居多？

huipengly, 2018-03-01:
4.x都破出来了啊，厉害。
编译看一下中文文档。
把帖子发出来吧，我有空的话可以帮忙编译。

aojie654, 2018-03-02:
@huipengly 没破解……就是做了一个拦截账号再拨号的so……
重庆so：http://right.com.cn/forum/thread-214612-1-1.html
拦截算法：http://right.com.cn/forum/thread-258825-1-1.html
# **_特别标注：以上内容来自恩山论坛_**

huipengly, 2018-03-02:
如果你所在地区运营商所用的是换行符也是\r\n，可以直接用帖子里编译好的so。

aojie654, 2018-03-03:
@huipengly 所在地陕西，不知道能用不，我回去测一下
另外就是也想学习一下这方面的东西😆

NorthNanmu, 2018-03-09:
你似乎理解错了帖子的意思(也有可能是我描述不清楚)
**4.x的算法从来就没破解出来!**
**4.x的算法从来就没破解出来!**
**4.x的算法从来就没破解出来!**
第一个帖子是我发的，当时由于4.x算法无法解析所以大家的解决方案都是无一例外的读取加密后的账号，但是麻烦的是拨号阶段路由器不支持\r\n（根本拨不出去），我把思路提供给我同学后(就是把github发给他看了一下= =)，他编译了删掉了@miao1007学长的拨号算法保留了\r\n的so文件，然后用C#写了一个从系统事件还是内存里读取加密账号的小软件，做出来后就我们俩人在用。当时simplenetkeeeper也能实现账号拦截，只不过要自己复制粘贴一下，我同学做的那个软件（应该也是参考了很多大佬的代码）不能发出去但是思路是通用的，于是就有了这篇帖子
第二个帖子中荒野无灯的拦截法重庆地区测试我也参与了[image]，这两个帖子中的解决方案都基于#138和[image]
所以4.x的算法就没有解出来过，**大家同根同源，本质都是一样的**

另外，如果你手头有k2p的话，刷了荒野无灯的固件后你甚至可以直接用winscp把荒野无灯编译的`magic.so`提取出来（我记得他讲的里面不会替换任何转义符，客户端发送的什么就拨什么），具体编译我也不会XD

附上荒野无灯的padavan其中的拨号日志：
```
Nov 24 11:28:06 pppd[22508]: Plugin rp-pppoe.so loaded.
Nov 24 11:28:06 pppd[22508]: RP-PPPoE plugin version 3.12 compiled against pppd 2.4.7
Nov 24 11:28:06 pppd[22508]: Plugin shanxun.so loaded.
Nov 24 11:28:06 pppd[22508]: sxplugin (author: miao1007, Mod: HuangYeWuDeng): init****
```

aojie654, 2018-03-09:
@NorthNanmu 没有啊…我知道没破解，大家都说n4加密很恶心这个我还是知道的…由于我是陕西的，所以就像学习一下看看这个陕西地区的so怎么编译我着手搞一个，我没说破解出来啦😜

aojie654, 2018-03-09:
@NorthNanmu 而且simple后来的版本已经不用复制粘贴了，直接打开一键连网就行…

aojie654, 2018-03-09:
@NorthNanmu 估计是simple内部包含了复制粘贴的方法吧