miao1007/Openwrt-NetKeeper #203: 关于服务器拦截法的一点可改进之处

shana321, 2018-03-03:
已成功使用服务器拦截法破解校园网，并且我还做了一个小白式教程发布到了网上。目前已有南昌netkeeper 重庆netkeeper 重庆天翼总共十多人破解成功（我已知的）。现在我有个大胆的想法，就是目前的这个方法我们只做了单向拦截：搭建服务器拦截拨号软件，缺少路由器作为客户端本身拦截服务器的功能。举个例子，最新版的扫码netkeeper，是需要不断从电信客户端获取信息以刷新二维码的，如果只有单向拦截，根本无法实现拨号，还有其他很多拨号软件早就是这种形式了。所以如果路由器能在拦截拨号软件前把服务器的信息转发给拨号软件，在拨号软件拨号的时候又能作为服务器拦截，最后再作为客户端向电信服务器拨号，实现双向的话，可以说全部的pppoe拨号软件都可以进行破解，要是能再加个心跳包就完美了。我目前不知道怎么实现，所以提出想法希望有大佬能试试！

shana321, 2018-03-03:
@huipengly 大佬你看你能实现这个吗，应该也是一个脚本的事

aojie654, 2018-03-03:
你发的教程就是我在#202里发的那个重庆so嘛？

shana321, 2018-03-03:
@aojie654 不是，so是miao学长发起的那个，我这个是分支

aojie654, 2018-03-04:
@shana321 我以为之前在恩山论坛发帖的哪个是你……

shana321, 2018-03-04:
@aojie654 那个是我啊，我是从这学的然后写了详细教程

aojie654, 2018-03-04:
@shana321 能不能麻烦大兄弟来一份编译so教程我拿去倒腾倒腾
学成归来辣条伺候😂

shana321, 2018-03-09:
@aojie654 emmm你好像说的是另一个帖子，之前没发现有这样的教程。至于so我没去研究，那个需要适合的文件 我不太想花太多精力所以就没弄了

left906080850, 2018-03-11:
[image]

left906080850, 2018-03-11:
[image]

shana321, 2018-03-11:
@left906080850 没懂，这个好像不是sh脚本吧？ 这个应该是心跳包源码吧

chain01, 2018-03-21:
@shana321 貌似手机不需要心跳验证，电脑会有心跳检测，

huipengly, 2018-03-24:
我的想法，拦截服务器发给客户端的不好拦。通信方式不知道，通信内容可能加密(甚至说一定加密)，拦截到了用处也不大。你可以自己试一试，或者看有没有大佬出来实现，试验你的想法了。这算是信全方面么？不懂。

shana321, 2018-03-24:
因为现在很多其他的拨号软件 以及新版netkeeper 在登录前都会获取服务器信息，用拦截法不解决这个问题的话会无法尝试拨号也就拦截不了了，现在有个新的容易实现的想法就是在拨号前实现内网穿透，拨号的时候再进行拦截，明月永在的固件似乎是实现了的 不过我暂未研究，linux学的不是很好  脚本自己也写不出来 只能看大佬了 @huipengly

huipengly, 2018-03-24:
我也不会。咱俩一起坐等大佬

hydejoe, 2018-03-27:
@shana321 
参照pppoe-relay

经测试可以转发内网的拨号请求以及后续的ppp包至外网，但是这种情况下只有该拨号的那台内网设备能连外网。（重庆，能拨号，未测试心跳）

可以尝试将内网其他设备发出的数据包先SNAT后加上pppoe包头，伪装成拨号设备的数据包。但问题一是这已经不仅仅需要编写脚本能力了，这应该已经算是底层驱动了，二是当数据包从外网发回时，是去掉pppoe包头发往内网其他设备，还是直接发回给内网拨号那台设备，那么这里就要维护一个table进行记录了，哦对还要应对“不同内网设备，但同源端口、目标地址和端口”这种情况。

解决上述问题后，只要拨号那台没断，内网其他设备都能连接外网

[man pppoe-relay](https://linux.die.net/man/8/pppoe-relay)

shana321, 2018-03-28:
好像是跟拦截法完全不同的原理，以我的理解暂且算是伪装法吧，但是这种方法需要解析数据包，我现在在想的就是改进拦截法 拦截法的优点是不管拨号加密算法如何更新都能破解。能否实现在pppoe拨号前pppoe-relay转发外网的包到内网，在路由器检测到有pppoe拨号请求时由pppoe-server进行拦截，这样一结合的话pppoe认证方式就已经完全破解了，不过还怕一个心跳包就是了。

shana321, 2019-01-05:
@huipengly 最近对Linux和openwrt了解了不少,支持VLAN的路由可以使用uci来将WAN桥接为LAN口。只需要这样改动脚本：将你nk4.sh中login失败的操作就设置为映射WAN为LAN，检测到用户名变化时除了执行再拨号外，在它之前加一个映射还原的脚本段，这样用户就能够扫码拨号了。不过手头没路由用了，3月份回学校看能不能写出来