miao1007/Openwrt-NetKeeper #217: 南昌的netkeeper今天更新了吗

DusKing1, 2018-11-02:
今天突然断线，再无论如何都无法成功拨号

`Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: Plugin rp-pppoe.so loaded.

Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: Plugin sxplugin.so loaded.
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: sxplugin : init
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: -------------------------------------
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: timenow(Hex)=%ßx
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: Begin : beforeMD5
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 1.<_@?>
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 2.<_@?------->
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 3.<_@?-------jiangxi4.0>
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 4.length=<24>
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: End : beforeMD5
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: Begin : afterMD5
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 1.MD5use_1=< 5>
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: 2.MD5use_2=< 1>
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: End : afterMD5
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: -------------------------------------
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: sxplugin : user  is <
1#0J<2 ----------------ndcard> 
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: sxplugin : passwd loaded
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: pppd 2.4.7 started by root, uid 0
Fri Nov  2 20:32:01 2018 cron.info crond[1631]: crond: USER root pid 19555 cmd sh /mnt/.init1.sh
Fri Nov  2 20:32:01 2018 cron.info crond[1631]: crond: USER root pid 19557 cmd sh /mnt/.init2.sh
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: PPP session is 30481
Fri Nov  2 20:32:01 2018 daemon.warn pppd[19549]: Connected to cc:1a:fa:e9:6c:40 via interface eth0.2
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: Using interface pppoe-net1
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: Connect: pppoe-net1 <--> eth0.2
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: syncppp not active
Fri Nov  2 20:32:01 2018 daemon.info pppd[19549]: CHAP authentication succeeded: Authentication Successful,Welcome!
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: CHAP authentication succeeded
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: peer from calling number CC:1A:FA:E9:6C:40 authorized
Fri Nov  2 20:32:01 2018 daemon.err miniupnpd[18167]: Failed to get IP for interface pppoe-net1
Fri Nov  2 20:32:01 2018 daemon.warn miniupnpd[18167]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Fri Nov  2 20:32:01 2018 daemon.err miniupnpd[18167]: Failed to get IP for interface pppoe-net1
Fri Nov  2 20:32:01 2018 daemon.warn miniupnpd[18167]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Fri Nov  2 20:32:01 2018 daemon.err miniupnpd[18167]: Failed to get IP for interface pppoe-net1
Fri Nov  2 20:32:01 2018 daemon.warn miniupnpd[18167]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: local  IP address ---
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: remote IP address ----
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: primary   DNS address ----
Fri Nov  2 20:32:01 2018 daemon.notice pppd[19549]: secondary DNS address ----
Fri Nov  2 20:32:02 2018 user.notice mwan3: ifdown interface net1 (unknown)
Fri Nov  2 20:32:02 2018 daemon.info pppd[19549]: LCP terminated by peer
Fri Nov  2 20:32:02 2018 daemon.info pppd[19549]: Connect time 0.0 minutes.
Fri Nov  2 20:32:02 2018 daemon.info pppd[19549]: Sent 0 bytes, received 0 bytes.
Fri Nov  2 20:32:02 2018 daemon.err miniupnpd[18167]: Failed to get IP for interface pppoe-net1
Fri Nov  2 20:32:02 2018 daemon.warn miniupnpd[18167]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Fri Nov  2 20:32:02 2018 daemon.notice netifd: Interface 'net1_6' is enabled
Fri Nov  2 20:32:02 2018 daemon.notice netifd: Interface 'net1' is now up
Fri Nov  2 20:32:02 2018 daemon.info dnsmasq[2674]: reading /tmp/resolv.conf.auto
Fri Nov  2 20:32:02 2018 daemon.info dnsmasq[2674]: using local addresses only for domain lan
Fri Nov  2 20:32:02 2018 daemon.info dnsmasq[2674]: using nameserver ----#53
Fri Nov  2 20:32:02 2018 daemon.info dnsmasq[2674]: using nameserver ----#53
Fri Nov  2 20:32:02 2018 daemon.notice netifd: Interface 'net1' has lost the connection
Fri Nov  2 20:32:02 2018 daemon.notice netifd: Interface 'net1_6' is disabled
Fri Nov  2 20:32:02 2018 daemon.warn dnsmasq[2674]: no servers found in /tmp/resolv.conf.auto, will retry
Fri Nov  2 20:32:02 2018 user.notice mwan3: ifdown interface net1 (unknown)
Fri Nov  2 20:32:05 2018 daemon.notice pppd[19549]: Connection terminated.
Fri Nov  2 20:32:05 2018 daemon.info pppd[19549]: Sent PADT
Fri Nov  2 20:32:05 2018 daemon.notice pppd[19549]: Modem hangup
Fri Nov  2 20:32:05 2018 daemon.info pppd[19549]: Exit.`

使用的是so没用拦截，网线直插电脑能用nk拨号且能成功，但是没有网络。
求教各位大大了

ChaoYihu, 2019-02-23:
同问,之前能连接但是被限速,而且隔一段时间会掉线.现在已经彻底连不上了.
网线连接电脑能够用NK正常拨号,怀疑是账号的加密方式改变了并且加入了新的心跳检测机制

xinxin8816, 2019-03-01:
哪个学校的？自拨号算法和拦截在南大都正常

xinxin8816, 2019-03-01:
楼主明显是没配置拨号用户名

ChaoYihu, 2019-03-01:
问题已经解决了,在用PC端的NetKeeper拨号之后再用路由器拨号恢复正常.
拨号算法在用wireshark抓包之后确认了确实没有问题.
谢谢大佬,同NCU

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

Humengchao, 2019-10-12:
> 问题已经解决了,在用PC端的NetKeeper拨号之后再用路由器拨号恢复正常. 拨号算法在用wireshark抓包之后确认了确实没有问题. 谢谢大佬,同NCU 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986>%E5%BA%94%E7%94%A8)
> […](#)

我也是ncu的，现在这个破解的还能破解学校的netkeeper吗？

ChaoYihu, 2019-10-12:
现在学校的电信网络已经不需要破解了 直接按照学校给的账号密码拨号即可上网

Humengchao, 2019-10-12:
难不成以前是需要买了账号还要破解的吗？可怕~~~  我还是这个学期开始用电信，感觉每次都用netkeeper不方便，而且一些物联网设备和linux系统设备就没法用了……，想搞成家里WiFi那样

ChaoYihu, 2019-10-12:
现在可以直接用了，就像家里一样。不过建议你在路由器上写个脚本  因为隔个两天左右会断网但不断线 用脚本检测断网后重连可以解决问题

获取 Outlook for Android<https://aka.ms/ghei36>

Humengchao, 2019-10-12:
不是吧，我每次连接还是需要netkeeper验证呀

 
 
 难不成以前是需要买了账号还要破解的吗？可怕~~~&amp;nbsp; 我还是这个学期开始用电信，感觉每次都用netkeeper不方便，而且一些物联网设备和linux系统设备就没法用了……，想搞成家里WiFi那样 
 
 
 
 ---原始邮件---

xinxin8816, 2019-10-13:
> 不是吧，我每次连接还是需要netkeeper验证呀
> […](#)
> ---原始邮件--- 发件人: "巢义虎"<notifications@github.com&gt; 发送时间: 2019年10月13日(星期日) 凌晨1:05 收件人: "miao1007/Openwrt-NetKeeper"<Openwrt-NetKeeper@noreply.github.com&gt;; 抄送: [...]

这个学期新办的电信宽带就不需要了，之前的还需要netkeeper。但是新的同时也不兼容电信千兆网提速接口了。取舍自己斟酌。推荐用旧的。

Humengchao, 2019-10-13:
不知道怎么搞的，我这个学期办的也是netkeeper的，🐶，那个github上的方法还有用吗？  还有千兆提速口是什么？

ChaoYihu, 2019-10-13:
用不用netkeeper都一样 你直接用Windows自带的拨号也是可以直接拨上的  路由器也一样

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

D

D

xinxin8816, 2019-10-13:
> 不知道怎么搞的，我这个学期办的也是netkeeper的，🐶，那个github上的方法还有用吗？  还有千兆提速口是什么？
> […](#)

这学期办的直接拨号就行了。GitHub上所有相关项目都无需用也不能用，包括千兆提速你不用想了。

Humengchao, 2019-10-13:
对的，我刚刚试了下可以在系统直接pppoe拨号上网，不用打开netkeeper，那这样的话是不是可以通过路由器pppoe拨号，然后其他设备直接连接就行了？

xinxin8816, 2019-10-13:
对的，直接用就行了

发送自我的 SAMSUNG GALAXY Note 10+

发件人: Humengchao
发送时间: 2019年10月13日 10:54
收件人: miao1007/Openwrt-NetKeeper
抄送: Zealer Chen; Comment
主题: Re: [miao1007/Openwrt-NetKeeper] 南昌的netkeeper今天更新了吗 (#217)

对的，我刚刚试了下可以在系统直接pppoe拨号上网，不用打开netkeeper，那这样的话是不是可以通过路由器pppoe拨号，然后其他设备直接连接就行了？

Humengchao, 2019-10-14:
学长，我这两天试了确实可以直接用。 还有个就是，这个能有公网ip吗？

 
 
 
 
 
 
 
 
 
 不知道怎么搞的，我这个学期办的也是netkeeper的，🐶，那个github上的方法还有用吗？  还有千兆提速口是什么？  
 …

ChaoYihu, 2019-10-14:
全国大多数地区的电信宽带都会随机分配公网IP地址的  NCU也不例外  如果你要在上面做服务的话 建议绑定动态域名

Humengchao, 2019-10-14:
南昌电信貌似可以直接找客服要，我之前也是为这公网IP才从移动转电信了，打电话过去，客服说给我弄好了，叫我重启试试，那时候还是没有。不知道是不是校园网的问题，

xinxin8816, 2019-10-14:
> 南昌电信貌似可以直接找客服要，我之前也是为这公网IP才从移动转电信了，打电话过去，客服说给我弄好了，叫我重启试试，那时候还是没有。不知道是不是校园网的问题，
> […](#)

电信手里握着的IPv4比移动联通加起来都多，而且校园网改造完后也是接入全国平台的。公网IP是肯定可以有的，没有就找客服要，还没有就多要几遍。如果你说的是IPv6那江西电信没有。

closed 2020-10-15 by DusKing1