miao1007/Openwrt-NetKeeper #229: x86 64位 pppd版本2.4.8的插件，需要自取

misterplus, 2020-09-29:
因为朋友需要编译的

链接：https://pan.baidu.com/s/1buuc0GGD9rPKs7qCKDDhmw 
提取码：l6yb

ChristianSwift, 2020-10-09:
我就是你说的那个朋友

misterplus, 2020-10-09:
> 我就是你说的那个朋友

草，朋友的朋友

wse320, 2021-04-10:
草，我也是你说的那个朋友，感谢

wse320, 2021-04-10:
草，我是arm的pppd版本2.4.8，原来的插件没法用了，楼主能不能帮个忙

djl0629, 2021-11-30:
> 

您好，我尝试编译x86 64的so文件时一直出错，想请教您更多关于编译的细节，实在是不知道错误原因，也找不到可以参考的资料
错误在进行make all操作时
makefile文件如下
#TODO:填写解压的Toolchain中bin与include所在的目录（根目录）,某些奇葩可能用的是glibc，这里可能要替换LIBC
LOCATION=../../openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl
#TODO:请注意不同型号(mips/mipsel)路由器此处gcc文件名可能略有不同
CC=$(LOCATION)/bin/.aarch64-openwrt-linux-gcc
#TODO:可能需要修改pppd的版本，取决于路由器的/usr/lib/pppd/，目前主流的14.09是2.4.7
PPPD_VER=2.4.8
CFLAGS=-Os -Werror -I$(LOCATION)/include -fPIC -DPPPOE_VER='"${PPPD_VER}"'

我在all中仅保留了我需要的杭州地区，默认全地区也有一样报错

$ make all
aarch64-openwrt-linux-gcc: warning: environment variable 'STAGING_DIR' not defined
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:139:3: error: #error
 # error
   ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:145:1: error: unknown type name '__STD_TYPE'
 __STD_TYPE __DEV_T_TYPE __dev_t; /* Type of device numbers.  */
 ^~~~~~~~~~
/usr/include/bits/types.h:145:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__dev_t'
 __STD_TYPE __DEV_T_TYPE __dev_t; /* Type of device numbers.  */
                         ^~~~~~~
/usr/include/bits/types.h:146:11: error: expected ';' before 'unsigned'
 __STD_TYPE __UID_T_TYPE __uid_t; /* Type of user identifications.  */
           ^
           ;
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/features.h:46,
                 from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/stdio.h:8,                 from sxplugin.c:1:
/usr/include/bits/types.h:146:25: error: 'uid_t' redeclared as different kind of symbol
 __STD_TYPE __UID_T_TYPE __uid_t; /* Type of user identifications.  */
                         ^~~~~~~
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/sys/types.h:57,
                 from pppd.h:55,
                 from sxplugin.c:6:
/home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/bits/alltypes.h:289:18: note: previous declaration of 'uid_t' was here
 typedef unsigned uid_t;
                  ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:147:11: error: expected ';' before 'unsigned'
 __STD_TYPE __GID_T_TYPE __gid_t; /* Type of group identifications.  */
           ^
           ;
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/features.h:46,
                 from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/stdio.h:8,                 from sxplugin.c:1:
/usr/include/bits/types.h:147:25: error: 'gid_t' redeclared as different kind of symbol
 __STD_TYPE __GID_T_TYPE __gid_t; /* Type of group identifications.  */
                         ^~~~~~~
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/sys/types.h:57,
                 from pppd.h:55,
                 from sxplugin.c:6:
/home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/bits/alltypes.h:294:18: note: previous declaration of 'gid_t' was here
 typedef unsigned gid_t;
                  ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
[... 148 lines cut ...]
 __STD_TYPE __SWORD_TYPE __intptr_t;
                         ^~~~~~~~~~
/usr/include/bits/types.h:209:11: error: expected ';' before 'unsigned'
 __STD_TYPE __U32_TYPE __socklen_t;
           ^
           ;
/usr/include/bits/types.h:222:1: error: unknown type name '__STD_TYPE'
 __STD_TYPE __TIME64_T_TYPE __time64_t;
 ^~~~~~~~~~
/usr/include/bits/types.h:222:28: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__time64_t'
 __STD_TYPE __TIME64_T_TYPE __time64_t;
                            ^~~~~~~~~~
In file included from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types/time_t.h:7:17: error: expected ';' before 'time_t'
 typedef __time_t time_t;
                 ^~~~~~~
                 ;
In file included from sxplugin.c:6:
pppd.h:231:20: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'groups'
 extern GIDSET_TYPE groups[NGROUPS_MAX]; /* groups the user is in */
                    ^~~~~~
make: *** [makefile:62: zhejiang_xiaoyuan_sxplugin] Error 1
$

djl0629, 2021-11-30:
您好，我在执行make all操作时遇到问题但是我实在没有办法解决，能不能请教一下编译中的细节，网上实在没什么资料可以参考

我使用的rockchip的openwrt21.0.2 sdk
我的makefile设置是这样的，我只保留了all中杭州的部分，未修改all中内容的时候也有一样的报错

#TODO:填写解压的Toolchain中bin与include所在的目录（根目录）,某些奇葩可能用的是glibc，这里可能要替换LIBC
LOCATION=../../openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl
#TODO:请注意不同型号(mips/mipsel)路由器此处gcc文件名可能略有不同
CC=$(LOCATION)/bin/aarch64-openwrt-linux-gcc
#TODO:可能需要修改pppd的版本，取决于路由器的/usr/lib/pppd/，目前主流的14.09是2.4.7
PPPD_VER=2.4.8
CFLAGS=-Os -Werror -I$(LOCATION)/include -fPIC -DPPPOE_VER='"${PPPD_VER}"'

这是我执行make all时的反馈信息

$ make all
aarch64-openwrt-linux-gcc: warning: environment variable 'STAGING_DIR' not defined
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:139:3: error: #error
 # error
   ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:145:1: error: unknown type name '__STD_TYPE'
 __STD_TYPE __DEV_T_TYPE __dev_t; /* Type of device numbers.  */
 ^~~~~~~~~~
/usr/include/bits/types.h:145:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__dev_t'
 __STD_TYPE __DEV_T_TYPE __dev_t; /* Type of device numbers.  */
                         ^~~~~~~
/usr/include/bits/types.h:146:11: error: expected ';' before 'unsigned'
 __STD_TYPE __UID_T_TYPE __uid_t; /* Type of user identifications.  */
           ^
           ;
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/features.h:46,
                 from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/stdio.h:8,                 from sxplugin.c:1:
/usr/include/bits/types.h:146:25: error: 'uid_t' redeclared as different kind of symbol
 __STD_TYPE __UID_T_TYPE __uid_t; /* Type of user identifications.  */
                         ^~~~~~~
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/sys/types.h:57,
                 from pppd.h:55,
                 from sxplugin.c:6:
/home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/bits/alltypes.h:289:18: note: previous declaration of 'uid_t' was here
 typedef unsigned uid_t;
                  ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types.h:147:11: error: expected ';' before 'unsigned'
 __STD_TYPE __GID_T_TYPE __gid_t; /* Type of group identifications.  */
           ^
           ;
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/features.h:46,
                 from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/stdio.h:8,                 from sxplugin.c:1:
/usr/include/bits/types.h:147:25: error: 'gid_t' redeclared as different kind of symbol
 __STD_TYPE __GID_T_TYPE __gid_t; /* Type of group identifications.  */
                         ^~~~~~~
In file included from /home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/sys/types.h:57,
                 from pppd.h:55,
                 from sxplugin.c:6:
/home/nnk/openwrt-sdk-21.02.0-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/bits/alltypes.h:294:18: note: previous declaration of 'gid_t' was here
 typedef unsigned gid_t;
                  ^~~~~
In file included from /usr/include/bits/types/time_t.h:4,
                 from /usr/include/net/ppp_defs.h:4,
[... 148 lines cut ...]
 __STD_TYPE __SWORD_TYPE __intptr_t;
                         ^~~~~~~~~~
/usr/include/bits/types.h:209:11: error: expected ';' before 'unsigned'
 __STD_TYPE __U32_TYPE __socklen_t;
           ^
           ;
/usr/include/bits/types.h:222:1: error: unknown type name '__STD_TYPE'
 __STD_TYPE __TIME64_T_TYPE __time64_t;
 ^~~~~~~~~~
/usr/include/bits/types.h:222:28: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__time64_t'
 __STD_TYPE __TIME64_T_TYPE __time64_t;
                            ^~~~~~~~~~
In file included from /usr/include/net/ppp_defs.h:4,
                 from pppd.h:57,
                 from sxplugin.c:6:
/usr/include/bits/types/time_t.h:7:17: error: expected ';' before 'time_t'
 typedef __time_t time_t;
                 ^~~~~~~
                 ;
In file included from sxplugin.c:6:
pppd.h:231:20: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'groups'
 extern GIDSET_TYPE groups[NGROUPS_MAX]; /* groups the user is in */
                    ^~~~~~
make: *** [makefile:62: zhejiang_xiaoyuan_sxplugin] Error 1
$

Starpuccino, 2022-03-29:
谢谢大佬、2.4.9又出来了，咱就是说能不能...😍

yuhanjin, 2022-05-30:
@djl0629     我也是纯小白。我和你的情况是一样的，用作者给的gcc可以编译，用最新的gcc就一堆一堆的报错。我感觉应该是c语言标准的问题。很多语法在老版本gcc上可以运行，在新版gcc上就当成语法错误报错了。毕竟作者给的是14年的gcc。
当然也不排除我哪里出了什么低级错误。

yuhanjin, 2022-05-30:
> @djl0629 我也是纯小白。我和你的情况是一样的，用作者给的gcc可以编译，用最新的gcc就一堆一堆的报错。我感觉应该是c语言标准的问题。很多语法在老版本gcc上可以运行，在新版gcc上就当成语法错误报错了。毕竟作者给的是14年的gcc。 当然也不排除我哪里出了什么低级错误。

@djl0629    我验证了我的说法是对的，我找了最新版的pppd.h文件，就能编译成功了
https://github.com/ppp-project/ppp/blob/master/pppd/pppd.h

weixieyou, 2023-03-06:
大佬，没有江西南昌的啊