miao1007/Openwrt-NetKeeper #234: 重庆nk遇到问题，家人们有没有遇到相同问题的

Milk-lpy, 2022-09-20:
Tue Sep 20 13:50:57 2022 daemon.err pppd[10717]: select (waitForPADO): Interrupted system call
Tue Sep 20 13:50:57 2022 daemon.warn pppd[10717]: Timeout waiting for PADO packets
Tue Sep 20 13:50:57 2022 daemon.err pppd[10717]: Unable to complete PPPoE Discovery
Tue Sep 20 13:50:57 2022 daemon.info pppd[10717]: Exit.
Tue Sep 20 13:50:57 2022 daemon.notice netifd: Interface 'netkeeper' is now down
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP ConfReq id=0x1 <mru 1480> <magic 0x696954de> <pcomp> <accomp> <callback CBCP>]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: sent [LCP ConfRej id=0x1 <pcomp> <accomp> <callback CBCP>]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP ConfReq id=0x2 <mru 1480> <magic 0x696954de>]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: sent [LCP ConfAck id=0x2 <mru 1480> <magic 0x696954de>]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: sent [LCP EchoReq id=0x0 magic=0xd5120f19]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP Ident id=0x3 magic=0x696954de "MSRASV5.20"]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP Ident id=0x4 magic=0x696954de "MSRAS-0-\302\344\273\250\310\313\262\273\303\302"]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP Ident id=0x5 magic=0x696954de "\3219[\341k-\356E\257Q\255u-k\273\352"]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [PAP AuthReq id=0x14 user="\r1/Y*,! b613320335362" password="143585"]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: sent [PAP AuthNak id=0x14 "Session started successfully"]

 Tue Sep 20 13:50:57 2022 daemon.warn pppd[10721]: PAP peer authentication failed for ^M1/Y*,! b613320335362
### 这个位置感觉不是很对头
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: sent [LCP TermReq id=0x2 "Authentication failed"]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP EchoRep id=0x0 magic=0x696954de]
Tue Sep 20 13:50:57 2022 daemon.debug pppd[10721]: rcvd [LCP TermAck id=0x2 "Authentication failed"]
Tue Sep 20 13:50:57 2022 daemon.notice pppd[10721]: Connection terminated.
Tue Sep 20 13:50:57 2022 daemon.info pppoe-server[7657]: Sent PADT
Tue Sep 20 13:50:57 2022 daemon.info pppd[10721]: Exit.

FBIx0024, 2022-10-26:
我也是这个问题，他偶尔出现，我是山东移动的

Milk-lpy, 2022-10-26:
我拨号625好像是

FBIx0024, 2022-10-26:
我就是正常报错691，然后现在拨号又好了，就是你得等会才能再次拨号，我举得报错的根源在于PAP peer authentication failed，这个导致的Authentication failed，然后我等了30min左右再次拨号，没有问题了